WATonomous/wato_monorepo #110: Implementation of HD Map (DRG)

Edwardius, 2024-05-31:


VishGit1234, 2024-10-15:
- HD Map
    - Generalize pedestrian regulatory element code to be for any obstacle
        - Mostly changing variable names
        - On the visualization side we have to display the obstacles as a 2d bounding box (convert polygon to marker array representation of a 2d bounding box)
        - Add attribute that stores the detected class (object hypothesis in the detection3d message)
            - Color code the bounding box based on the class
    - Visualization of traffic sign and traffic light
        - Store traffic light state (or sign type) as attribute in reg elem
        - These can be simple lines (color coded for the traffic light state)
    - Write code for traffic sign reg elems following similar pattern as pedestrian reg elem

JStephenHuang, 2025-05-24:
Here is the HD Map todo list to complete the implementation: 

https://github.com/orgs/WATonomous/projects/21/views/1?pane=issue&itemId=95388764&issue=WATonomous%7Cwato_monorepo%7C170