WATonomous/wato_monorepo #111: Implementation of Localization Node

Edwardius, 2024-05-31:


VishGit1234, 2024-10-29:
GTSAM Approach: 
- Being worked on by Harsh

Localization using robot_localization:
- Work off this branch (https://github.com/WATonomous/wato_monorepo/tree/world-modeling-robot-localization)
- Add robot_localization as a package
- See here for what the architecture looks like (https://docs.ros.org/en/melodic/api/robot_localization/html/integrating_gps.html#using-navsat-transform-node)
- Mainly using nodes from the robot_localization package so just creating correct config and launch files

JStephenHuang, 2025-06-05:
For Summer 2025, let's just get this thing rolling!

I made a detailed doc on the localization. It should give a good idea of what we are trying todo. It's a rough draft, so let me know if I should I add anything or correct anything.

[Localization.pdf](https://github.com/user-attachments/files/20602136/Localization.pdf)

Of cource, we still have to make docs to show the team how to run and how to use the localization node and output accurate (x,y,theta) of the car.

I made concrete sub issues under this ticket to implement the localization node!