WATonomous/wato_monorepo #114: Costmap Modification from Predicted Paths

Edwardius, 2024-05-31:


VishGit1234, 2025-03-06:
Input: 

- [Detections3DArray](https://github.com/ros-perception/vision_msgs/blob/ros2/vision_msgs/msg/Detection3DArray.msg)
   - Each of these detections represents an object on the road, (could be a car or pedestrian)
- Velocities for each of the detections (for now this can mocked to some constant)
- In the future the above 2 inputs will probably just be one input being a composite message

Output:

- Costmap (same costmap topic) where for each detection the bounding box is projected forward based on the velocity for some constant horizon (this should be a parameter) and these cells are filled in the costmap with some intensity lower than the max intensity

Where to code: Directly modify occupancy_core.cpp, adding helper functions for the relevant computations

VishGit1234, 2025-03-26:
Inputs:
BoundingBox3D msg -> car 
Twist msg -> velocity
Above two come from somewhere (don't know yet)
OccupancyGrid msg -> occupancy (the current occupancy code)

Output:
OccupancyGrid msg -> same occupancy (costmap) as input but for the bounding box we project its shape in the direction of the velocity for some fixed horizon (maybe 1 second?)   

Work in same package (occupancy) and create files for new node