WATonomous/wato_monorepo #125: (HIGH PRIORITY) 3D Object Tracking

danielrhuynh, 2024-06-14:
We need 3D object tracks to send to world modelling
Potential Implementation: 
https://github.com/ZHOUYI1023/awesome-radar-perception?tab=readme-ov-file#Velocity-Estimation

**June 13 2024**
- Have tracked objects working in Nuscenes!
- Working on integrating actual detections from the perceptions pipeline (blocked by 2D to 3D(

danielrhuynh, 2024-06-28:
**June 27 (Skipped a week for midterms)**
- Steven's node not working properly
- Kishore reimplemented AB3DMOT in a new ROS node which functions properly with synthetic data? Still need to test with nuscenes?

danielrhuynh, 2024-07-05:
**July 4**
- Outputs look good
- Can't integrate with 2D to 3D yet since it doesn't build but we need the node for the specific message type
- Building the visualizer node locally
- Integrate into a pipeline soon and do e2e testing

danielrhuynh, 2024-07-12:
**July 11 2024**
- CAM bag was recorded for testing 2D to 3D
- We need to record a bag for CAM to get 2D detection arrays for testing 3D DET node
- Visualizer will be done in the next few days
- Will be wrapped up soon!

danielrhuynh, 2024-07-26:
**July 25 2024**
- Core node is done just needs to be tested

Kishore-Yogaraj, 2024-09-24:
- Tasks in progress for 3D Tracking:
- Record ros bag of 2D to 3D converter
- Create visuals with opencv
- Test tracking with visuals on ros bag

Kishore-Yogaraj, 2024-11-12:
- Meshva's Ros bag was unavailable so testing was put on hold
- New ros bag recording was put on hold because of an error with QoS profiles (fixed
- A new ros bag with 2d detections will be recorded by EOD today
- Build error with Meshva's 2d to 3d node (dependency issue as of now)
- 3D markers are currently not working directly with foxglove
- Attempting to use the message converter extension shown here: https://foxglove.dev/blog/using-message-converters-to-display-3d-markers-in-foxglove-studio to display 3D markers

Kishore-Yogaraj, 2024-11-18:
- 2d to 3d node needs more attention or we need to switch to monocular depth estimation instead
- 3D bounding boxes aren't being draw accurately
- Converted nuscenes data set to ros bag instead to continue testing of tracking node
- Working on creating a config for processing nuscenes data
- Will be using visualization_msgs/msg/MarkerArray as the message type to visualize 3d tracks in foxglove

Kishore-Yogaraj, 2024-11-26:
- Need to play around with transforms and adjust accordingly to visualize 3d bounding detections with 2d_3d node
- 3D bounding boxes are now drawn using visualization_msgs/msg/MarkerArray with Nuscenes data set
- Node has been configured to handle nuscenes data for tracking out
- Need to work with transforms from map view to base link as tracked objects are appearing as stationary in some cases
- Should also implement an option to track with OpenCV rather than nuscenes and publish the annotated image