WATonomous/wato_monorepo #175: Sensor Integration/Testing

Kishore-Yogaraj, 2025-02-01:


lucasreljic, 2025-02-23:
- Seems that the Eve computer was updated to Ubuntu 24 which is causing issues with running Spinview in terminal it says something along the lines of "shared package could not be found"
- All 3 cameras were temporarily attached to the car and connected to the ethernet switch
- Lidar also connected to the ethernet switch with power

Future tasks:
- Install docker and the monorepo
- Install camera driver in a docker container with ros2
- Either downgrade to Ubuntu 22 to fix Spinview or try to get cameras running with driver directly or find different fix for Spinview

lucasreljic, 2025-03-02:
- Docker was installed along with the monorepo
- It seems that Ubuntu 22 is partitioned on the drive so you just need to switch to that partition
- Spinview was able to view all the cameras

Future tasks:
- Install camera driver
- View LIDAR data and install the LIDAR driver

lucasreljic, 2025-03-07:
- Got 1 camera publishing to ros

Future tasks:
- Get all 3 cameras publishing to ros
- Get LIDAR data
- Install new LIDARs and integrate them into system

lucasreljic, 2025-03-19:
- Got LiDAR ROS node publishing
- Continued debugging multicamera node getting "Acquisition timeout" error

thomasonzhou, 2025-09-28:
In progress:
- static IP assignment (preferred over DHCP)
- PTP timestamping (IEEE 1588)
- resolve frame drop for 3 cameras (possible culprits include debayering on CPU)

https://github.com/ros-drivers/flir_camera_driver/blob/humble-devel/spinnaker_camera_driver/doc/index.rst
https://docs.ros.org/en/humble/p/spinnaker_camera_driver/

closed 2026-01-09 by brianzheng206