WATonomous/wato_monorepo #184: Fix bugs in Camera Object Detection

lucasreljic, 2025-03-18:
Fix small bugs in camera object detection such as:

- Change the path in the docker-compose yaml file to the correct model for traffic_signs (traffic_signs_v0)
- Fix the image width and height error
- Either fix the camera encodings on Eve or implement a method in camera object detection to convert encoding to the correct type

brianzheng206, 2025-04-20:
Fixed CUDA buffer memory crash issue; TensorRT engine was previously built and deserialized each batch callback without freeing cuda buffers

Optimization to do's: 
- CLEANUP PARAMS (constructor has way too many params being inited and not used)
- use onnxs nms library, currently uses ultralytics nms on the CPU every frame batch (single threaded)
- switch to int8 or fp16 inference (currently fp32) **this would half mem use**
- lower publishing rate of the viz or just get rid of it entirely, only used for debugging
- offload image preprocessing in the gpu; cv2 cuda resize and cuda cvtColor onto GPU

brianzheng206, 2025-05-10:
batched inference node detection publish smoothly without visualizations; still has room for optimization

wm needs class_id's from object detections to be published with the /batched_detection_msg, currently publishing indexes of classes and not the class names themselves

closed 2026-01-18 by Edwardius