WATonomous/wato_monorepo #186: Sensor Interfacing (Cameras/LIDAR) node

lucasreljic, 2025-04-03:
Create a new docker container and ROS node in interfacing for the cameras and the LIDAR.

brianzheng206, 2025-04-14:
Working lidar and camera ros drivers in docker containers in the stgong/sensor_drivers branch, need to make a launch file to launch multiple spinnaker cameras, the current ones only launch one at a time

brianzheng206, 2025-04-15:
all 3 cameras have working ros drivers, however there's some bandwidth issues causing the frames to drop
- look into docker settings
- adjust parameters of the camera config, the order of the parameters are crucial and must be correct for some settings as specified in the documentation

brianzheng206, 2025-04-21:
frames now publish at a constant 24-28hz with around 4-7% drops the most consistent I could get it to run
working lidar ros driver and 3 camera ros driver

to do:
need to calibrate the 3 cameras: https://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration
the debayering node doesn't work; images are being published as BayerRG8, either figure out how to process the image and remap the image_raw topic or leave it for post-processing